BAMresearch/amworkflow #10: add create geometry by points

aradermacher, 2023-10-13:
add function for creating geometry by points in geometry/composites_geometries.py and in the GeometryCenterline class. Testing via example Truss or new simpler one (as in Yuxiang's branch)

aradermacher, 2023-11-16:
Hi Yuxiang, I saw that the trussarc example works now in this branch. Can we merge the branch into the main?

aradermacher, 2023-11-16:
some comments:

- I agree to use is_close as global parameter for the GeometryCenterline Class.
- In geomtery.py I would like to import the CreateWallByPointsUpdate as "from amworkflow.geometry.composite_geometries import CreateWallByPointsUpdate " and call add in this way in the GeometryCenterline Class
-

aradermacher, 2023-11-16:
On my vdi the example tooks quite a while, is that normal

aradermacher, 2023-11-16:
Last comment: I checked the design and it looks like there are two holes missing:
OLD
[image]
NEW:
[image]

YuXHe15, 2023-11-16:
Hi Annika, For now it is barely a simple replication from the branch v0.43 on which I haven't implemented the improvements of the new CAD module. I can finish the integration by next week.

I was actively changing parameters, trying to solve the problem when bending geometries, which may cause the error of missing holes. I ll look into this issue and fix it.

Therefore can we wait for several days for my updates?

YuXHe15, 2023-11-16:
I think The long processing time is due to I disabled the multithreads in Gmsh in a previous Fix, because it is not compatible with Doit multithreads. Otherwise it should be faster considerably

closed 2024-02-02 by aradermacher