BAMresearch/amworkflow #20: Create G-Code module

YuXHe15, 2023-11-08:
Create a module designated for G-code parsing and generating.

aradermacher, 2023-11-09:
There was already an issue for that #12. 
But it's fine, I will delete #12.

aradermacher, 2023-11-16:
I would prefer to have a similar gcode class as for the geometry and the meshing.

Something like:
class Gcode:
    """Base class with API for any gcode writer."""

    def __init__(self, *args, **kwargs) -> None:
        self.logger = logging.getLogger(__name__ + "." + self.__class__.__name__)

    @typing.override
    def create(
        self, in_file: Path, out_gcode: Path ) -> None:
        """Create gcode file by given path file or geometry file

        Args:
            in_file: File path to path point file or stl file from geometry step
            out_gcode File path of output gcode file.

        Returns:

        """
        raise NotImplementedError

class GcodeFromPoints(Gcode):
     "Gcode generater from given path points"

    def __init__(...)
       super().__init__(**kwargs)

    def create(self, point_file: Path, out_gcode: Path):
         "write gcode by given points"
        like the code from yout temp_convertor.py

YuXHe15, 2023-11-16:
Hi Annika, I completely agree with your idea. I am working on some fundamental functions and classes which works for the final Gcode class.

My plan is to create a parametrized gcode parsing and generating module, working for machines with various gcode standard.

The thing is that neither iso-6983 nor DIN-66025 define a detailed code norm. The Reprap firmware does have a detailed documentation but there are also several versions like Marlin, repetier,etc, which only share most commands in the same codes.

The printer I was working on yesterday is installed a reprap firmware, but the version maybe Repetier.  Therefore for example the code M104 which supposed to control extruder temperature actually controls the extruder switch. 

Therefore I think it would be a useful part of the code.

[image]

The structure looks more or less like the figure presents.

closed 2023-12-13 by YuXHe15