BAMresearch/amworkflow #30: Add functions for multiplying gcode output

YuXHe15, 2024-01-16:
**To solve a practical issue**: The printer with which we are working can only perform in its best when extruding continuously. 

**Solution**: Therefore to print multiple structures while sufficiently utilizing the printing palate, a good way would be to output gcode files with various original points,  which are defined by a grid partitioned the palate.

aradermacher, 2024-01-17:
I agree with this issue. We just have to discuss the interfaces for that task. I would like trying to stick to the interfaces we have. Maybe a new step "merging gcode" with single gcode files as input could work?

YuXHe15, 2024-01-23:
One more thing. I also add a function for output logs for gcode files. If you run the code you would find a series of csv files which contain details of the printing process, which proved to be useful for the material preparation. Since we need to continuously provide fresh materials during the printing, it is crucial to figure out how much materials will be consumed during 15 minutes and this is where these csv files work.

aradermacher, 2024-01-24:
First of all I cannot run your example, because I don't have the "in_file_path" .

I would like to stick to the idea that each "functionality" can be used individual by a set of required parameters and files.
Would it be possible to rewrite the class by using a gcode file as input insteadt of gernating the gcode part in the class by using the GcodeFromPoints class. You would need to extract the "body" of the gcode for the points and use those in the gcodelist variable.

Then the workflow would look like that:
gcode = GcodeFromPoints(gcode_params)
gcode.create(csv_point_file, single_gcode_file)

mulgcode = GcodeMultiplier(params) # params are here the one you used already without gcodeparams plus auto_balance, dist_horizont, dist_vertic
mulgcode.create(single_gcode_file, gcode_file)

In this way you could do that also for gcode produced with some commerical or free slicer software.

If this is not possible, I prefer to add the new class as alternative gcode creater as child of GcodeFromPoints, then you can use the function from that class directly.

What do you think?

YuXHe15, 2024-01-24:
> First of all I cannot run your example, because I don't have the "in_file_path" .
>
> I would like to stick to the idea that each "functionality" can be used individual by a set of required parameters [...]

That is certainly possible. If I understand it correctly, it would require at least a very simple gcode reader right? After that I can extract points from one gcode file and replicate.

I think it is because I deleted one demo csv file which contained points, sorry for that. It is in the /usecases now and I hope you can run the code now.